adept-anyball/mod #5: Check Aliexpress BTUs for compatibility

efogdev, 2025-02-18:
Link: [https://vi.aliexpress.com/item/1005007460320866.html](https://vi.aliexpress.com/item/1005007460320866.html)

efogdev, 2025-02-18:
I'm currently waiting for them to arrive.

efogdev, 2025-02-27:
SP-8 nylon ball: https://github.com/user-attachments/assets/d8cfdc46-2cf6-4129-a199-c57a8b72ef77
SP-8 steel ball: https://github.com/user-attachments/assets/b1ad3f8a-16da-45e3-a90e-de85919935fa
Rexroth for comparison: https://github.com/user-attachments/assets/21eb35c5-aafb-43b8-8942-3b341e5ec625

benmccown, 2025-03-06:
Just got my SP-8s in the mail today. Going to try printing the alternate top, will update on how it goes.

I watched the videos you shared. As someone who doesn't have Rexroth BTUs to compare against, how big of a quality difference do you notice?

HuyLeKouta, 2025-03-07:
I have seen the showcase

So the SP-8 BTU can be used in the newer 50-55mm shell ? 

Thanks for the re-confirmation

efogdev, 2025-03-07:
@benmccown speaking of the steel ball version, it's honestly almost the same. I'd say Rexroth still feels better in long run, maybe just more stable? Not sure.

@HuyLeKouta yes

benmccown, 2025-03-07:
@efogdev So I printed the SP8 compatible top and I'm trying to use it with my 52.5mm snooker ball, but it seems to be sitting way too high off the sensor to detect any mouse movement. It looks like mine sits a good bit higher than the example videos you've attached in the original issue comment. 

Any ideas if I've set anything up incorrectly? What ball size did you use in that original example?

[image]

[image]

benmccown, 2025-03-07:
@efogdev Got it fixed up. The issue was the sensor cover along with me not realizing how hard you have to push the BTUs into the sockets to seat them. I had to get some sandpaper out so I could make more room to actually push them down.

efogdev, 2025-03-07:
@benmccown glad you got it working! Yes, the tolerances are really tight, but it should be like that, because it would be really loud otherwise.

efogdev, 2025-03-07:
Also, I assume you need the sensor cover? I'll make it soon and will update you

efogdev, 2025-03-08:
Actually I just realized you probably can grab the sensor cover from the "57-68mm ball" version, and it will probably be fine.

benmccown, 2025-03-08:
@efogdev I actually just used the sensor cover from the "Everything" file for the 50-55mm. Maybe that's why it was so tight?

efogdev, 2025-03-08:
Yeah it's not supposed to be used with that cover, I'd recommend to print the one I've mentioned above. I probably should make it more clear in the README.

JayBee-git, 2025-03-28:
Sadly the 57-68mm sensor cover doesn't really work, it's way too small so it's just floating there moving around, not attached to anything.
I got another weird issue with the SP8 top, I fully inserted the BTUs, but somehow the ball doesn't go right to the bottom, the left BTU looks like it's too high making the left gap bigger. Might just be a tolerance issue, but visually both BTU look completely inserted.
@efogdev Would it be alright to share a STEP file with us so I can tweak it a little? Thanks a ton.

efogdev, 2025-03-29:
Damn, I didn't know I forgot to upload a STEP for this one. The thing is, I also forgot to make a checkpoint in the CAD :D
I will probably have to make this one again (not that it make much time but still), I'll ping you then

benmccown, 2025-04-29:
@efogdev I went to print the updated SP8 top housing and it looks like you have the sensor cover included in the file and not as a separate body. Unfortunately the print failed because the sensor cover tried to print with the body and the overhang angle is too high.

efogdev, 2025-04-29:
@benmccown yep that's right, I just thought it'll be a mess if I start to include sensor covers and everything else as separate files. You're supposed to split those in your slicer (the same applies to the main model), I'm not sure how to make it more clear in README

benmccown, 2025-04-29:
@efogdev got it, thanks mate. sorry for the confusion. I just got a new printer (Bambu P1S) and so it's my first time using the Bambu slicer, and the file shows up as one body so I was confused and thought it wasn't splittable. Let me do some more learning with the slicer and see if I can split it.

j1n6, 2025-05-17:
@efogdev possible to make a thin/short version of this?

efogdev, 2025-05-17:
@j1n6 possible but would take fair amount of work.

j1n6, 2025-05-17:
> @j1n6 possible but would take fair amount of work.

Hope you may have sometime to work on it - this is slightly outside my knowledge...