adept-anyball/mod #7: Gallery

efogdev, 2025-02-19:
You're welcome to show your Anyball here!

efogdev, 2025-02-20:
50-52.5mm version vs 50-57mm side-by-side comparison.

[image]

NayteDog, 2025-03-01:
57-68mm mod with a pool ball
[image]

benmccown, 2025-03-07:
50-55mm with SP8 BTU compatible top housing, with a 52.5mm snooker ball. I've combined it with [the AdeptBLE](https://github.com/taichan1113/AdeptBLE) project so it's bluetooth running ZMK. I'm going to have to reprint the top housing because I had some warping issues during printing that don't impact functionality, but will also bother me as a perfectionist :). [Build details on Reddit here](https://www.reddit.com/r/ploopy/comments/1j62mwi/wireless_bluetooth_ploopy_adept_adeptble_with_a/). More pics [here](https://imgur.com/a/TYVjxkV).

[image]

pandrr, 2025-03-08:
45mm ball / rexroth BTU

[image]

pingjocky, 2025-03-15:
57-68mm mod with a pool ball

[image]

yuhua99, 2025-04-24:
[image]
works beautifully with sp8 btu, great job @efogdev

zodduska, 2025-07-19:
55mm short thin bezel in FDM .12 PLA

[image]
[image]