adept-anyball/mod #10: Tolerance issues for BTU sockets and magnet sockets

benmccown, 2025-03-07:
Hey there! With my recent build, using the SP8 top housing 50-55mm, I couldn't get the BTUs to fit without sanding each hole for 3-5 minutes each. I went to superglue my 4x2mm magnets and ran into the same problem- they wouldn't fit. I just superglued the buttons to the top instead of using magnets.

I do plan on reprinting the top housing sometime soon because I had print warping issues. Going to tune my printer up and try again. But I'm wondering if there's any way we could get a 0.1mm or 0.2mm tolerance bump for the areas I've mentioned? Especially the magnets, since they don't really need to be perfectly tight if you're using superglue anyway.

Everything was printed on an Ender 3 using standard PLA, 205 C nozzle temp, 0.2mm layer height, and default Prusa slicer settings for the Ender 3. Since I'm running a cheaper printer I'm guessing that may be the issue, but I'm sure others out there will want to build this project and might not have a top of the line Prusa or something.

Thanks for an awesome project.

Cheers,
Ben

efogdev, 2025-03-08:
Hey, thanks for the feedback. You're not the first one mentioning that problem, so yeah, I will eventually rework all the shells. 

Actually (for those who maybe waiting), that's pretty easy to do with 3 clicks in OnShape — you just import the STEP, use "move face" tool with like 0.1-0.2mm value, then apply and export.

benmccown, 2025-03-08:
I will give it a shot! Thanks for the tip. Will give you an update when I try it (probably in a few days).

efogdev, 2025-03-08:
I just tried that and apparently it's not the case for this model :( "move face" doesn't work with this particular face because of... reasons.

But I of course have the full features history so won't be an issue. I'll update you in this ticket in several days, ping me in case I forget :)

fabriciobastian, 2025-03-08:
I would also suggest that you run a calibration on your printer, if you haven't done so. The magnet holes have a .2 tolerance, which is a full layer. It should be enough for them to at least fit tightly.
You could also check whether there are seams on the holes and if you can print without putting seams on the magnet holes.
Otherwise, I guess that we could add another .1mm tolerance to the hole diameters @efogdev.

efogdev, 2025-03-09:
@fabriciobastian I remember us talking about making it 4.35mm :)

fabriciobastian, 2025-03-10:
@efogdev that works too :)

JayBee-git, 2025-03-26:
Yeah I got the same issue, most 4x2 magnets we can find are 4.15~4.2mm and not exactly 4, meaning even with a well calibrated printer, the tolerance is still too tight.
It sadly a common issue with these magnets. I had to increase the size of mine to 4.5 with a rotary tool.
Indeed I'd love models with 4.35~4.5mm holes too :)

efogdev, 2025-03-26:
I'm too excited to work on [the wirelessifier](https://github.com/efogdev/wirelessifier) because I got the rev.02 PCBs on hands yesterday 😅 

But I promise I'll do the changes once my internal hype goes away!

JayBee-git, 2025-03-27:
It's alright I completely understand. Do you have a step file or similar? I could do it on my side if you want and share the file, not an issue for me 👍

drewofdoom, 2026-09-03:
For everyone struggling with this, here's a simple solution that worked for me: https://github.com/supermerill/SuperSlicer/wiki/Polyholes

Polyhole conversion is an option in my slicer (Anycubic Slicer Next).

The top needed some slight scraping, but the buttons were perfect right off the bed. The holes were too small on a previous attempt without the conversion.