adept-anyball/mod #15: BOM for AdeptBLE PCB Assembly

leclaron, 2025-03-21:
Here is the BOM for getting the PCB for the AdeptBLE assembled from factories like PCBway and JLCPCB, I have picked components that are possibly the cheapest options available from suppliers like Mouser and Digikey.
The only thing not included in this BOM is the Seeeduino Xiao NRF as it isn't possible to have it soldered to the PCB at the factory. 

[AdeptBLE BOM.ods](https://github.com/user-attachments/files/19398146/AdeptBLE.BOM.ods)

efogdev, 2025-03-22:
Hey, I'm not sure how this is relevant to Anyball; anyways, I think it makes sense to create not an issue but a discussion

efogdev, 2025-03-22:
I've nothing against this info being in this repo, it's useful for Adept owners. But it would be great if you could create a discussion instead (issues isn't quite the right place IMO)

closed 2025-03-22 by leclaron